visualapproach/Volvo-melbus #2: Audio ground loop?

visualapproach, 2017-03-24:
Data transmissions is audible when not playing. I have audio and data running in the same CAT5 network cable.

visualapproach, 2017-03-26:
It appears to be the arduino that is the source of the noise. (It goes silent when resetting the arduino. Remains if I disconnect the audio, wich suggests it is NOT a ground loop problem.) I'll try decoupling the power lines with a cap.

sorano, 2017-03-26:
Nice find! 

Såg precis att du uppdaterade fotot, Lol @"fritzing" ;) 

schematic.png visar pins 8, 11,12 för BT medans foyot & koden använder 8,9 & 12.
Ska se om det är mycket på jobbet imorgon annars kan jag börja klippa och klistra lite riktig fritzing :) som komplement till fotot.

Är det bakelit pcb du använder?

visualapproach, 2017-03-27:
Tack. Tyvärr hjälpte det inte med en konding. Angående pin-val så spelar det ingen större roll vad gäller att kontrollera BT-modulen (Däremot måste data-pinnarna hamna rätt för de använder interrupt som bara finns på vissa pinnar). Du bestämmer funktion genom att definiera pinnarna i början på koden.
Jag hade ett gammalt experimentkort liggande. Tror det är glasfiber eller liknande. Kopparn är rader om tre hål:
--- --- --- --- ---
--- --- --- --- ---
osv, men det går så klart med andra typer också.

visualapproach, 2017-03-28:
Glömde säga det, men det vore ju jättebra om du gjorde en riktig fritz. Kanske blir enklare för nästa person som vill vara med. Jag har nu åkt till jobbet några dagar och lyssnat på BT-musik. Riktigt nice! Ibland kopplar inte iphonen upp sig automatiskt mot modulen utan man får klicka på "anslut", men sen funkar det bra. Modulen spelar en liten plingplong när den ansluter.

sorano, 2017-03-28:
Har börjat skissa lite i Fritzing nu, ge mig nån dag så ska jag ha det någorlunda ok!

Vet du om BT modulen klarar av [aptX](https://en.wikipedia.org/wiki/AptX )? Har märkt av skillnad i ljudkvalité på BT jämfört med AUX?

visualapproach, 2017-03-29:
Jag har faktiskt ingen aning om aptX. Man måste nog kolla datasheetet för själva chipet och det är lite krångligt när den är i bilen för det behövs i princip ett mikroskop. Jag hör ingen skillnad mellan AUX och BT. Det är ju subjektivt men jag har inget att klaga på vad gäller ljudkvalitén. Jag är nöjd.

mrkiller2010, 2017-03-29:
Haha, sjukt hur jag hittat er som fortfarande kör runt dessa volvos! :) Har en Volvo V40 2001 fas1 med en HU radio.
Köpte en 8pin kabel till aux hona. Fick hem det idag och har inte hunnit att testa det, så måste man göra en interface och använda sig utav kod för att få auxen att funka? :/

Sorry för offtopic

visualapproach, 2017-03-29:
@mrkiller2010 
Hej och kul att du hittade oss ;-)
enligt [denna sida](http://forums.swedespeed.com/showthread.php?133953-Guide-for-swapping-or-upgrading-MELBUS-head-units-radios-in-your-2000-2009-P2-Volvo) så har du en HU som använder MELBUS och dessa måste man lura att det finns en CD-växlare för att de ska öppna LINE-IN. Finns komersiella grejer att köpa, eller så har man inte vuxit ur "kansjälv-stadiet" och värmer upp lödpennan.
Det är ok med off topic, det finns ju ingen allmän chat man kan skriva i vad jag har sett.

mrkiller2010, 2017-03-29:
@visualapproach 
Okej, nu är jag med. Men din lösning har även support för BT.
Skulle du kunna länka alla delar som krävs för denna interface? Finns ju alternativ till audrino, skulle det funka bättre?

Det [http://i.ebayimg.com/images/g/DKEAAOxym2BSGzav/s-l300.jpg](här) jag köpte. Fast med AUX hona så jag kan borra i den till själva radion

visualapproach, 2017-03-29:
@mrkiller2010 
Först skulle jag testa bara sladden du köpt. Vem vet, kanske har den inbyggd elektronik. Fast jag skulle mäta på den först så du inte kör 12v rakt in i mobilen eller vad du tänkt koppla in. En av pinnarna på DIN-kontakten är nämligen 12v.
Om du ska bygga så läs [readme](https://github.com/visualapproach/Volvo-melbus/blob/master/README.md). Om du inte vill ha BT så ta bort följande komponenter:

3x 560Ω
3x LED's (To drop voltage and look cool)
BT Audio Module

Koppla därefter in ljudsladden till resp pinne på DIN-kontakten. [Länk till DIN pin layout](http://volvo.wot.lv/wiki/doku.php?id=melbus)
Ladda upp koden på Arduinon och så ska det fungera.

Angående Arduino: För att aktivera LINE-IN genom att emulera CD-C så fungerar Arduino (i motsats till vad vissa säger). Det finns så klart en massa andra alternativ, men det är till Arduino jag gjort programmet. Om det nu är Arduinon som orsakar det lilla störljudet innan man aktiverar ljudspelning så kanske det funkar bättre med nå't annat men det har jag inte provat.

visualapproach, 2017-03-30:
@sorano 
Fick igång SAT som SOURCE. Det kan innebära en del bra saker... 
[readme](https://github.com/visualapproach/Volvo-melbus/blob/master/README.md)

sorano, 2017-03-30:
Coolt 😮 !

Vad står SAT för i detta fallet? Jag antar Satellit då det står Sirius? Varför vill man ens ha tillgång till satellit i sin bil? För satellit telefon? Eller TV om man har skärmar?

Hoppas du hittar något intressant 😃

visualapproach, 2017-04-01:
I got rid of 90% of the noise by driving the AUX (on the BT module) to GND via two resistors (Between 100 and 600 Ohms would be fine. I tried with 240 Ohms)
red---resistor---black---resistor---white
Good enough for me, so case closed.

closed 2017-04-01 by visualapproach

mrkiller2010, 2017-04-30:
Undrar om du kan löda ihop allt om man beställer hem alla delar hem till dig för en slant?

visualapproach, 2017-05-07:
@mrkiller2010 Det blir inte lönsamt för någon av oss. Skillnaden mellan en kommersiell produkt och denna är bara några hundralappar som ska täcka både jobb och frakt.

mrkiller2010, 2017-05-09:
@visualapproach Hejsan!
Har precist beställt allting som du skrivit på listan, var tvungen och köpa flera stycken på vissa som resistor, LED, och dom där små sakerna.

Skulle du kunna skicka flera bilder på bygget? Vill se hur allt är lödat fast. tack

mrkiller2010, 2017-05-09:
@visualapproach Förresten, vad heter den där gula plattan allt sitter på? lär ju beställa en sådan också?

visualapproach, 2017-05-10:
@mrkiller2010 Det är en experimentkort. Det finns massor olika modeller och det spelar ingen större roll vilken du köper. Det finns billiga i papper med koppar på baksidan, men de är lite jobbigare att löda. Sen finns det lite dyrare i bakelit/glasfiber etc. Sök på "prototype experiment pcb" så hittar du massor. Ta en modell med 0.1" hålavststånd (2.54 mm) och som har tre-fyra hål per "ö". Dvs kopparn förbinder tex tre och tre hål. Då är det lättare att löda fler kablar till samma PIN osv. [exempel](http://www.ebay.com/itm/10Pcs-DIY-Prototype-Paper-PCB-Universal-Experiment-Matrix-Circuit-Board-8-5x20cm-/131047274580?hash=item1e8306b854:g:a-AAAOSwgApW~3~j)
Angående bilder på bygget så har jag monterat allt i en plastlåda som är jobbig att skruva isär. Är det nåt specifikt du vill veta så kan jag nog förklara.

visualapproach, 2017-05-10:
Ett tips när du bygger ihop allt: gör en anslutning så att du kan ta loss modulen utan att behöva demontera bilradiopanelen varje gång. Som du ser på mina bilder har jag ett antal små kontakter (kräver specialtång för att crimpa dit på kablarna), men man kan också skarva nätverkskabeln med tex en hanDIN+honDIN.

mrkiller2010, 2017-05-10:
@visualapproach Hejsan. köpte det du länkade precis.
Det jag undrar är hur allt är lödat, typ vilka kablar osv men det löser sig nog när jag får hem allting. 
Detta är enbart bluetooth eller hur? Såg att du skrev att man kan enkelt koppla aux till blueetooth enheten och fixa det på det sättet? 

Och vad menar du med att fixa en anslutning? Jag ska ju löda ihop allting och slänga in allt i en liten låda med sladd ut till själva MEL Bus porten till radion?

visualapproach, 2017-05-11:
@mrkiller2010 Hej, AUX ska bara vara att plugga in med medföljande sladd. (Minikontakt i BT-modulen som slutar med lösa kabeländar, så du kan koppla på valfri AUX-kontakt.) Med anslutning menar jag att skapa ett enkelt sätt att koppla loss plastlådan. Säg att du måste löda om något, då är det enklare att koppla loss boxen och gå och löda, än att riva ut stereon för att koppla loss den. Och bättre än att sitta i bilen och löda ;-)
En annan grej du kan tänka på (som står i schematics) är att jag blev av med det sista oljudet genom att INTE koppla in AUDIO GND. Lycka till med byggandet!

mrkiller2010, 2017-05-12:
@visualapproach Tackar för tipsen. Har du lyckats fixa custom text? Skulle vara coolt om man kunde ändra så att det står låtens titel eller BT/AUX på radion.

visualapproach, 2017-05-17:
Det vore jättecoolt men jag har inte forskat något mer. Det krävs nog att man sniffar en satellitradio för att lyckas. Lite svårt att gissa sig fram.

mrkiller2010, 2017-07-05:
Hejsan!
Har fått hem allting och jag har börjat löda ihop modulen.
Är inte riktigt klar men undrar vart två tråds sladden som sitter i BT chipet ska lödas fast samt. 

Tror dessutom att jag lött helt felt men vi får se..  :/

visualapproach, 2017-07-05:
Kul! Du ser på bilden i readme vart den ska sitta i bt-modulen. Observera fel färgkod på sladdarna, så svart=+12v och röd = gnd. I alla fall är det så på min. Andra änden: svart direkt på Vcc(12v) och röd (gnd) på transistorns kollektor. Lägg upp ett foto eller skiss här om du vill ha stöd innan du kopplar in.

mrkiller2010, 2017-07-05:
Är trött på skiten haha. 
Har du lust att hjälpa till med lödningen? är i stort sett klar, bara nätverkskabeln till 8pins anslutning samt till själv arduino och bt chipet kvar.

mrkiller2010, 2017-07-05:
Såhär långt har jag kommit, http://imgur.com/a/F5gko Det är säkert helt fel komponenter jag använt samt lödat fel haha.
Ska försöka löda ihop resten imorgon och se om skiten fungerar vilket jag tvekar på.

mrkiller2010, 2017-07-06:
Fan det är omöjligt o få nätverks sladdarna på 8pins skiten, råka smälta skiten så pinsen e sne nu :/
Har en till 8pins connector men vill ej göra den då jag inte kan. Sen så fattar jag ej hur jag ska ansluta nätverkskabeln till själva bt modulen o det. 

har du inte lust o hjälpa mig om jag skickar allting hem till dig?

visualapproach, 2017-07-06:
Var bor du?

mrkiller2010, 2017-07-06:
Bor i Gävle

visualapproach, 2017-07-06:
Ok lite för långt. Jag jobbar men skriver mer ikväll. Som tröst kan jag säga att jag också smälte den första kontakten. Man får vara snabb och inte ha för hög värme eller för stor lödspets. Det bästa är om man har den inkopplad i en hona så stiften sitter fast medan man löder.

mrkiller2010, 2017-07-06:
Hur många timmar ifrån mig bor du om man åker med bil?

visualapproach, 2017-07-06:
6-7 skulle jag tro

mrkiller2010, 2017-07-06:
Kan annars skicka hem allting till dig, och du hjälper mig med det som är kvar? Står för porto och allting.

visualapproach, 2017-07-06:
Hann tyvärr inte kika på detta ikväll men ska göra det så fort jag får tid.

visualapproach, 2017-07-08:
@mrkiller2010 vilken email har du?

mrkiller2010, 2017-07-08:
@visualapproach dyar.98@hotmail.com