visualapproach/Volvo-melbus #3: HU is commanding power on repeatedly

visualapproach, 2017-03-24:
This isn't a problem, but annoying when sniffing the data line.

sorano, 2017-03-30:
Jag funderade lite på detta.
Kan det inte vara så att HU vill ha någon typ av "ack" acknowledgement som verifierar att enheten den söker är igång, och då slutar skicka HU skicka "power on"?

visualapproach, 2017-03-31:
Jo det är helt riktigt. Själva "powerup"-meddelandet ska ackas med en byte. Jag har provat alla svar (0-255) utan att märka skillnad. Tror att man måste hitta rätt i antingen CartridgeInfo (6 bytes) eller i TrackInfo (9 bytes). Det är dock lite jobbigare att testa brute force på så många bytes. (6 bytes = 256^6 = 281 biljoner kombinationer. Trackinfo är lite färre okända bytes.) Men när jag orkar tänkte jag göra ett mer interaktivt program så jag inte behöver kompilera om för varje ny kod, och så kan man kanske få fram betydelsen av fler bytes.

visualapproach, 2017-03-31:
Actually, I found an answer here on github. Maybe. I'll have to check...

visualapproach, 2017-04-01:
The answer was found here: [code](https://gist.github.com/klalle/1ae1bfec5e2506918a3f89492180565e)
Case closed

closed 2017-04-01 by visualapproach