visualapproach/Volvo-melbus #6: AUX & BlueTooth?

sorano, 2017-03-27:
What would be the best course of action if you want to use an AUX connector along with BlueTooth?

Connect AUX to the BlueTooth module or piggyback the cables to DIN?

visualapproach, 2017-03-27:
I'd connect it to the AUX on the BT module. It's supposed to automatically switch to AUX input when BT is quite (or not connected. The "manual" is sparse to say at least.) There is also a MIC input on the board, that I guess could be used to speak on the cellphone. I have a VOLVO corded Telephone installed so I haven't tested it :-D

sorano, 2017-03-27:
Will try and report once I get my module.

closed 2017-03-27 by sorano

mrkiller2010, 2017-05-10:
@sorano Hello, Did you manage to get AUX input working as well?