visualapproach/Volvo-melbus #10: chat

visualapproach, 2017-07-11:
put your comments or questions here

visualapproach, 2017-07-12:
Check the cables extra carefully that comes with the BT Module! Red and black power wires may or may not be reversed. I've seen both...

rysatko, 2017-10-17:
Hello Thomas.
First of all I would like to thank you for your work. This is awesome project. 
I tried to make my own dongle (for now only AUX-in, CSR8645 is on the way from china). But I managed to work it only once - both SAT and CD-CHGR. Then it showed SAT ERROR and since that time my HU850 (with 6CD built in) does not recognize it. I tried all your SW versions and also SW from Vincent, but no luck. Could you please go through the troubleshooting with me? When I open serial monitor, it tells only "Calling HU" and then plain lines only. After uploading Vincents SW it tells "not in sync".
HW side should be fine. I measured all connections several times and everything is OK.

visualapproach, 2017-10-17:
Hi and thank you for joining! 
My first guess would be the power supply. The car electric power is very noisy and that can cause problems. I had some communication errors too when I switched from using the RUN wire as power supply to using 12v from the cigarette lighter. It was solved by NOT using the cig lighter as a GND-point, but using the shielded cable's shield. See picture below. The reason I didn't use RUN is that it couldn't deliver enough amps for my project. At least not consistantly. In the pic below I haven't included protection against backpowering. It can be done with a resistor and a transistor (see master thread) or several other ways. I really recommend the ground loop isolator also. It is worth the money (I think 20 €-ish)!

[image]

rysatko, 2017-10-17:
For now my HW is built exactly iaw the drawing in master thread. So I am taking +12V from RUN and it is grounded by pin #2. Plus I added 8205 voltage regulator because the regulator on my chinese Arduino seems a bit crappy and I measured only 2V on the 5V pin. So I hope that the solution is so easy as you suggest. I will let you know asap. :)
Btw. ground loop isolator is also on the way.

visualapproach, 2017-10-17:
Fingers crossed :-)
I think you could try to power it with a powerbank via USB. (Easier than resoldering stuff for a test)

I wonder if it's the voltage regulator or if RUN is not coping...

rysatko, 2017-10-17:
So it is connected to the lighter socket. But no change for now. I am taking my computer to the car and we will see..

rysatko, 2017-10-17:
Nothing again. HU does not recognize it. :(

visualapproach, 2017-10-17:
Ok... did you try driving the arduino from the computer ONLY? (Disconnect 12v cig and 12v  RUN, and connected GND through DIN cable)

visualapproach, 2017-10-17:
Do you get any text except "calling HU"? If not then I suspect the main init sequence is not performed successfully. Maybe it doesn't pull the busy line low enough? And confirm your arduino is 5v 16MHz. The 3.3v 8 MHz won't do. It might be helpful if you post some pics on your device.

rysatko, 2017-10-17:
I am sure, that it is 5V arduino. As I told you, it worked for 2 days. I can see text "calling HU" in serial monitor. Now I am going to desolder GND and take it to car again, this time powered from powerbank only.

visualapproach, 2017-10-17:
Ok. It can be really frustrating and take a lot of time finding what's causing the trouble. But since it has worked something must have happened. Which means it is not likely software. So either something broke physically or electronically. Several times I have been certain it was a bug in my program, but it always ended up beeing a loose wire, or bad ground or similar. It's difficult to say what's wrong in your case and all I can offer at this point is some generic recommendations:
-check voltages and resistances with a multimeter (like is 12v OK even when under load? Same for 5v. Is GND resistance to HU chassi almost 0 Ohms? Recheck DIN connector, it's prone to bending and in a tight space there is a risk of shorts.
-make another visual inspection of the cables. Look for shorts or open circuits in the cables and under/over the PCB.
-do not use extremely long wires/cables.
-when in doubt, reduce to simplest possible circuit and check functionality. (only arduino, powered from battery)
-get a couple of beers and sleep on it. Repeat this step if necessary.

rysatko, 2017-10-17:
The last step would be the best :D 
No succes with power only from USB. Only "conn" or "calling HU" messages.
To go through hints you gave me:
- Resistances checked, everything is fine. Voltages not yet, only on table when powering it from 12V PC power source - was ok. Bending of DIN is not my case, because I am trying it with HU pulled from dashboard.
- I have done inspection for shorts with ohm meter - ok
- Longest wire is approx 0,5m cable to DIN and 1,5m audio cable (does not matter)

Maybe I can try to buy another Arduino and try it again.

When I think about it, I have bought AUX/USB/Bluetooth box from china and I also had issues with it. Sometimes it worked, but mostly not. This was why I wanted to build my own. Maybe my HU constantly has its days. :/

visualapproach, 2017-10-17:
Yes maybe it's the HU, I have a 650 myself. No play in the HU connector? If you wiggle it I mean. Anyways, you should see almost 5 v on the data/busy/clk-lines at the arduino side when you measure, and 12 v on RUN. If you don't, you know the problem is either your HU or the cable. If possible you should measure this first with arduino disconnected (because it pulls the lines high) If this is too cumbersome you can upload a sketch that tristates D2-D4 (pinMode(2, INPUT))
I really wish you good luck now, and keep us updated on your progress!

rysatko, 2017-10-17:
I will try the code with input pins on D2-D4, because Arduino is soldered to prototype board and it would be pain in ass to get it off. I can also build second cable, but it is nonsense. I will try it on Thursday (if I will not be in work till night :D ). 
Also the Arduino could be bad. As I told you, the voltage converter on it is quite tricky and when I powered it directly from 12V, the LEDs on it lit only slightly. That time when I powered it with 5V from NTB or via 8205, it worked like a charm. 

Thanks so much for your time and kindness. I really appriciate it. ;)

rysatko, 2017-10-28:
Hello again.
I finally tried it. I'm not at home and I forgot to take USB cable to upload modified software so I tried to disconnect GND cable from the whole box (in my opinion it should work too, but maybe I am wrong). I have got 3V on D2 and D4 and 4,5V on D3.

visualapproach, 2017-10-28:
Hi again! Hang on and I'll check what my DIN output measures at the end of the cable...

visualapproach, 2017-10-28:
Yes, As I suspected I measured 4.7 volts at all three lines (In car, arduino end of cable, disconnected from arduino. I have an extra pair of DIN plugs so I can easily remove the device from the car). So I think you've traced it to be either your cable, or your HU. Next step would be to remove the cable and measure on the HU itself.

rysatko, 2017-11-04:
Finally a day off - HU measured. RUN - 11,26V, DATA and CLK - 4,57V, BUSY - approx. 4,5V (fluctuating between 4,42V and 4,58V. Now I will try to desolder the cable from my board and try it again. The cable is 8x0,25mm2, connector is standard solderable 8-pin DIN 270°.

visualapproach, 2017-11-04:
Good! It is a good thing that your HU is healthy. It is much easier to fix the cable. I use a shielded network cable with 4 twisted pairs in it. I had to run GND through the shield, rather than the GND at the cig connector, to get rid of alternator noise. (See image in this thread). I wish you luck!

rysatko, 2017-11-04:
Connector checked and soldering is rock solid. So I desoldered BUSY, CLK and DATA at the side of Arduino and measured once again. And I found out, that when I move with the cable at the HU side, voltage of BUSY is droping to 0V. This is a hope :)

visualapproach, 2017-11-04:
Yay! Bad wire? Great find! Intermittent faults are tricky to hunt down.

rysatko, 2017-11-04:
I changed the wire aaand... Nothing again. :/ It is still losing connection. I think the problem is in the connector. But even if I tried to hold the cable in position where it appeared to be OK, nothing happened. HU still refuses to cooperate. I think I will get another Arduino and I will give it one more try...

visualapproach, 2017-11-04:
Is there a point in getting a new Arduino if you know your cable/connector is bad? Of course it doesn't hurt getting some extra arduinos, they're pretty cheap.
Or do you get good readings out of the cable when it's not connected at the arduino side now?

rysatko, 2017-11-04:
I have connection, when the cable goes straight forward or when I push it towards the drivers side. I tried to connect everything together and HU stil does not see any new device. For now I know the voltage converter on Arduino is bad. It is few years old and was used for several projects. It cost me about 3€. Maybe there could be also problem with the 8205...
I will build new board without the backpower protection and we will see. Just a question - what is the capacitor C2 at the drawing you posted at the beginning? I ve got only C1 - 470uF

visualapproach, 2017-11-04:
Oh, c2 is a small ceramic cap. I put it there with the purpose of filtering noise when I had some issues. You can omit it. The big cap can be considered more of a local powerbank to survive short voltagedrops.

rysatko, 2017-11-11:
I'm starting getting frustrated. Today I gave it a try. New Arduino, new cable and new simple board. When I was measuring connections between HU and board, I found out, that I do not have voltage at CLK pin. Even if I try to measure it directly in the HU connector, I cannot see anything...

visualapproach, 2017-11-12:
That sucks! I can't really give you any advice on how to fix that. Sorry. But I'm feeling your frustration!

rysatko, 2017-11-12:
Now I am just thinking... I have only connected DATA, BUSY, CLK and GND lines into HU. Pins for right and left audio and GND on pin are free. Have you tried this setup, when you were developing your board? Maybe HU shuts off CLK, when some of those 3 are free? I did not try to connect any load to them when I was measuring this time. And I do not remember, if I connected them when I was measuring the time before.

I am just cooking from water, before I get courage and time to remove HU from car and start disassembling it. :D

visualapproach, 2017-11-13:
@rysatko I measured 5v on DATA, BUSY and CLK directly on the HU connector. Nothing connected at all.

PedroPiko, 2017-11-21:
Hello Thomas,

First of all I would like to thank you for your work. This is very interesting also to me. I have tested on HU-603, which has not SAT mode. So I changed the CDC to MDC base, response IDs and now I running with MDC (buttons not tested on Nano outputs) but except text seems that is running OK. I have debug it with serial and found that MBR2 is there but there is no text in MDC. Do you have any idea how to enable it: Thank you. Peter

visualapproach, 2017-11-21:
Hi Peter!
Thank you. My 650 doesn't support MDC, I think. So I have no means to investigate it by my own. @VincentGijsen and I managed to 'crack' the SAT messages between his 3rd party commercial device and the HU by using a logic analyzer and a lot of testing. I have a logic analyzer but nothing to analyze. If he reads this he MIGHT be able to sniff some more? 
My guess is MDC messages are equal or less complicated to understand than SAT though.
To enable MDC text you would need to:
- record and analyze communication between a MDC (real or emulated) and HU. (Theoretically one could try to answer the HU without this recording, but it would take too much effort and time. Number of expected bytes in the answer is known, but not the expected content.)
- Add support for new commands from HU, along with proper answers.

Regarding the MRB2: I reckon you are requesting master mode as MDC and the HU grants it by sending out a master request broadcast to every device connected on the Melbus network. You then have to listen for MDC ID and respond with ResponsID. That's the easy part. Immediately after that you have to send some data. Most likely different from CDC and SAT, so you can't just send data as were it a SAT device.

Cheers / Thomas

VincentGijsen, 2017-11-21:
Hi Peter,

As Thomas said, we sniffed using a hu850 (popup screen in my case) .

As the aftermarket unit supposed to use SAT-mode with this radio, I sniffed only that. I guess the mechanism will be similar on MD, but as the sat has perhaps more options, it might have a more complex initialization mode. Perhaps when i have some spare time in the coming weeks, I might put the MD firmware on the unit and see if I can record a initialization. But I don’t know if the 850 will query the device at all.

Cheers

visualapproach, 2017-11-21:
Hi Vincent! Nice to hear from you again. Very kind of you to reply. Hope you have a good time!

PedroPiko, 2017-11-21:
Thomas, Vincent,

Thank you for info and support. I have also logic analyser, but nothing to analyze too 👎.I will try to find something on the net.

Peter

visualapproach, 2017-11-21:
Some guys asked for MD info on this page: [https://gizmosnack.blogspot.se/2015/11/aux-in-volvo-hu-xxxx-radio.html](url)
Perhaps they found something out by now and are willing to tell you. And then you post your info here ofc :-)

PedroPiko, 2017-11-22:
Thomas,

I have started with that project (gizmosnack) and ended up with this. I went trough all the related projects and have found some guys that have found solution for text in MDC mode. Most of them turn to commercial so they would not to share the code. Anyway I will try to carry out some tests.

Peter

visualapproach, 2017-11-22:
Yeah I thought so but it was worth mentioning. If you or vincent gets any data logged you are welcome to post it here for people to analyze. If it is large amount of text please start another issue to keep this chat human readable.

erinel01, 2018-02-11:
Hi!
I went with your old schematics, ver 2 but i get + from the lighter socket and ground from the shield of the DIN, signal-ground I take from pin 2 in the din.
I have a ticking sound that is the same rythm as the led on the arduino. It´s present regardless if I have the BT-module connected or not. Anyone have a suggestion what to look for?

visualapproach, 2018-02-11:
Hi,
I never got rid of that until I switched to CSR BT module. At the same time I also installed a ground loop isolator. Never tried that on previous bt module. The ticking was only heard at high volume during paused music though. I think it’s crosstalk in the cable between the data lines and audio lines. Now that i’m thinking about it my audio lines is going through the GLI outside of the DIN cable, joining at the connectors. Don’t know if that routing makes any difference. But my system is free from noise now.

VincentGijsen, 2018-02-11:
A groundloop isolator does help alot, also i wiredup the Ground from the din to powersupply Ground.

Furthermore i had Some issues with with poweramp on the csr board, which tripped some brownout threshold during ignition from an ams1117 regulator, so I switched the csr board via an transistor after a second. Steady as a rock :)

Btw the CSR is totally awesome in terms of audio quality, but unfortunately no ACRP 1.3 (uart and metadata)

johan489, 2018-04-23:
Hi,
Quick question, what type of groundloop isolator are you using?

Cheers

johan489, 2018-04-23:
> Hi @johan489 <https://github.com/johan489>
> I bought this one from the local electronics store: link
> <https://www.kjell.com/se/sortiment/ljud-bild/kablar-adaptrar/rca-audio/signalomvandlare/avstorningsfilter-3-5-mm-p39757>
> The reason I chose that one was because I could get it the same day and it
> was kind of cheap. Pick one with suitable connectors for your build.

Thanks,
Starting assembly soon, waiting for a few parts.

j-ak-e, 2018-06-13:
Hey @visualapproach I just discovered your project when reading https://gizmosnack.blogspot.com/2015/11/aux-in-volvo-hu-xxxx-radio.html

Did you ever get custom text working? Could I help with custom text at all?

visualapproach, 2018-06-14:
@j-ak-e yes sending text to HU display is working fine :-)
It's all in the code, with comments.

visualapproach, 2018-06-14:
@j-ak-e that is, if your HU can select SAT mode. For MD mode no work is done. Needs data sniffing from a real or 3rd party device.

j-ak-e, 2018-06-14:
@visualapproach I get my V70 in a couple weeks, so I'll make a breakout cable for the accessory port and sit in the car debugging, I'd be able to sniff data but I don't have access to a device unfortunately

visualapproach, 2018-06-14:
@j-ak-e Me neither.

j-ak-e, 2018-06-20:
@visualapproach I'm trying to lay out a PCB to make a small batch of these (~10), did you ever find part numbers for the DIN sockets/plugs?

johan489, 2018-06-20:
I would be interested in buying one if you do.

Johan

VincentGijsen, 2018-06-20:
Will you ADD a Bluetooth module, ifso, i Can recommend an csr module. Great sound quality. But if you pick a slightly more expensive module, you can can also get the metadata over serial. And write that to screen.

Kind regards,

Vincent

Sent from a mobile device

visualapproach, 2018-06-20:
@j-ak-e 

just google "DIN plug, 8 pin, 270 degrees". The cheaper ones requires more soldering skills because the plastic easily melts. If you buy some females, solder while male-female is joined, to secure the pins in place.

j-ak-e, 2018-06-20:
@visualapproach It was for the PCB layout, I've found a couple of working parts, so I'll use them, currently waiting on my EDA guys to design the footprint, and some money to come in to order boards

dancek, 2018-06-24:
Hi! Just wanted to say thanks for sharing your code. I soldered a simple cable (with 3.5mm plug) for my HU-803 and this code  works nicely.

I'm thinking about porting the relevant code to Rust on ARM for practice. That is if I find enough motivation to figure out all the voltage level conversions...

visualapproach, 2018-06-25:
@dancek I appreciate that you share the success with us! Well done! I know nothing about rust, but I think you have a lot of room for prettyfying the code with that kind of horsepower. From the top of my head, there should only be three lines to level convert. Busy, data and clock.

j-ak-e, 2018-06-25:
@visualapproach (or anyone else with a module that has in-built power regulation), could you figure out what voltage I should be supplying the chip, and on which pin, if not using a module with regulator? Do I provide 1V8, 3V3 or 5V to VBUS?

visualapproach, 2018-06-25:
@dancek which chip are we talking about?

j-ak-e, 2018-06-25:
@visualapproach assuming that was directed at me, the CSR8645

visualapproach, 2018-06-25:
You assumed right ;)
There are several variants of breakout modules. Mine was with a 12v input IIRC. Some appears to require 3.6-6.5V. If you are using a bare chip i don’t know. A google search suggested max 3.3 V but you’ll have to research that. Or wait for a more educated answer.

j-ak-e, 2018-06-25:
I'm planning to use this: https://www.banggood.com/CSR8645-4_0-Bluetooth-Stereo-Audio-Module-Low-Power-Consumption-Supports-APTX-p-1104609.html but the only specs they provide are for the battery. If you get chance, could you measure the voltage between VBUS and GND on your module @visualapproach ?

VincentGijsen, 2018-06-26:
I use such a module without  12v part. I just added a ams1117 3.3v regulator. The amp on the board has some issues powering up sometimes, so I connected it’s power via npn and digital io , so I can reset the board and do a delayed start.

If you spend a few bucks more, you can buy something like https://www.silabs.com/products/wireless/bluetooth/bluetooth-classic-modules/wt32-bluetooth-audio-module

Way more Bluetooth control and metadata access :)

Kind regards,

Vincent

Sent from a mobile device

[114 more replies]