visualapproach/Volvo-melbus #13: Suggestion regarding the BT

VincentGijsen, 2017-08-09:
I would use a CSR8645 (its twice the price with headphone amp). but sounds so much better, oh and it supports APT-x and AAC, as well as realy advanced tuning it's noice-canceling, bi-microphones and equalizer. But straight out of the box, it's as easy as your parallel controlling wires.

Next step should be a BT with serial-interface and AVRCP and meta-data. but they start at 20+

visualapproach, 2017-08-10:
Thanks for the info! Good to have options. A search on ebay for "CSR8645" came up with some modules from $6 which is in the same pricerange as my module.

VincentGijsen, 2017-09-03:
regarding text and bluetooth, you need one of the more expensive units, bluegiga, w32i, or melody sw. They start at >25+ and supply an uart interface. the heavy lifting of the avrcp and metadata is handled within these customized csr modules (basically they wrote specifik sw, running in the VM section of the csr-chips). If you want a real challenge, you could take a look at the ADK from csr, but it's not easy to come by, Some Chinese forums seem to have it (if you pay with credits). but I guess it's not as trivial as just buying some of the repackaged/flashed options with an easy uart interface

visualapproach, 2017-09-09:
Ok. I was playing with an xs3868 and it has avrcp "control only". I could see some potential benefits though, in hooking up the arduino via RX/TX to the chip. For example you get the current state of the music: playing/paused. Controlling the PLAY input PIN is only toggling so you must know the starting conditions if you want to know whether the music is playing or not. I also managed to toast the onboard voltage regulators on that chip. That was easy. Perhaps my 8 channel levelconverter sucked the guts out of it :D

JamieEC, 2021-03-25:
Hi All, did you have any luck with a BC127? Or any other of these modules for that matter? I am tempted to buy one as the support looks better than the others despite the price. I want to add fast-forward and rewind options, as well as some hands free phone capabilities