visualapproach/Volvo-melbus #14: Power supply

visualapproach, 2017-08-13:
It appears the RUN line (providing 12V DC) is not particularly powerful! I got some weird chopping in the sound sometimes. Now I am feeding 12 v from the cig lighter output and no more hickups. Instead I have alternator wining sound, but that should be fixable :-)

VincentGijsen, 2017-08-14:
the alternating sound, is a result to your probable ground-loop. Just buy a (cheap) groundloop isolator, and all your problems are solved. place it inbetween your audio path, this separates the audio signals galvanic.

I put in a small bc547 transistor inbetween the ground and the audio circuit. insert a xKohm resitor to the base, and wire that to the RUN wire, this will turnoff your hardware when the radio is turned off (which protects the battery)

visualapproach, 2017-08-14:
Yes I gotta try that and see if it helps. I might need a big Cap or a coil on supply also. And I'll use the transistor solution there as well (as in the main GND on my schem.) Good point. A shielded cable wouldn't hurt either.
By the way, I'm working in parallel with you on udpating the software according to the info you gave me about the SAT comm. Really glad you came along!

VincentGijsen, 2017-08-14:
Well, i wil incorporate an orange pi (with analogue video) with the rti screen, and would like to use that as a video/audio player. 

Regarding the Cap it will probably make for a cleaner signal, but the the bobine/coil will stay present unless you add a groundloop filter (only few bucks on ebay).

Any progress thus far on the text?  I've also noticed that when i put my transmission inreverse, the screen comes up, enabling the (not connected) rear camera. so that's a nice gimick to add as well, just 2 pins on your RTI to add the camera to.

visualapproach, 2017-08-14:
> Any progress thus far on the text?
Not really. I was working on it but I couldn't even get it into SAT mode tonight. And I thought I had a good chance to see bitches with the updated code... (anyone reading this should see Vincent's code to understand what I'm talking about)

One peculiar thing is that in your sniffdump the device responds with 0xC5 in main init. I have only had any success with 0xC6. Can you verify that?

VincentGijsen, 2017-08-15:
Mm you might be onto something.

The dump lists indeed C5 as response to baseID C0. My code responds differently. Which might need correction. I guess I copied that byte form some forum, instead of the dump.

The funny thing though . The code currently in my test file. Does seem to initialize sat partly correct. That is, no error. Although I hear some noice repeating in the speaker, and see retries of a certain command. And no text so far.

Are you also using hu850?

visualapproach, 2017-08-15:
@VincentGijsen I'm on a HU650. No RTI or popup screens. I'll stick to C6 when HU asks for C0. It was late when I tried this yesterday and I found some errors this morning which might explain why it didn't work. After merging your code with my own I ended up disabling INT1 instead of INT0. Doh! We have different hw layout so note to self: copy new code to own project, not the other way around.
Note to you: when device is in master mode, you need to provide your own clock!
The timing info in your dump file is very useful! (After converting it to delta time)

VincentGijsen, 2017-08-15:
Hahaha, I all forgot about the clock generation in master mode... that would definitely help. LOL!

And indeed, my pins are slightly different and therefore the interrupt .

I'll try after work

visualapproach, 2017-08-15:
I got sat mode to work again. I found a logic shortcoming in the code. I'll post an update but I first want to get my mastermode not to lock up...
You mind sharing your sniffing hw setup? Do you piggyback the pc onto imiv/hu?

visualapproach, 2017-08-17:
Alternator sound SOLVED. It was a bad ground connector.

mrkiller2010, 2017-08-28:
Great that you've solved the issue, by feeding the cigarette lighter do you mean connect a phone charger for example or?

visualapproach, 2017-08-29:
@mrkiller2010 Instead of taking 12v from RUN you take it from somewhere else. I made a Y-splitter from the contact behind the cig outlet. Not visible from outside and no chargers needed.

I also tried another BT module (the cheap xs-3868) and it is not as sensitive for noise on the 12v as the module in the pictures. It has other drawbacks though. I'm also waiting for another module to test (see Vincent's suggestion) and hope to get crystal clear sound :-)

2bitmarv, 2019-09-16:
Measuring the voltage supply in my 2006 V70 II shows 14.4V when engine is running. Arduino is working flawless but the BT module with its own 12V isolated supply seems to be burned up since this "ND 1203S-1W" DC/DC converter on the PCB only digests ~12.8V at max. I assume that because short before i mounted the HU-650 i did a final test on my desk with my straight 12V power supply and it worked. After assembling the HU in to my car, the Arduino works fine, but the BT never shows up. 

Is the circuit handling this mighty supply? Or does this circuit expect the supply to be almost around 12V?

visualapproach, 2019-09-16:
From memory: the diode from 12v should drop .7V but googling ND B1203S-1W they state input range to be 10-20V. I never had problems with that but there is an option to get a non regulated module and use an external voltage regulator.

visualapproach, 2019-09-16:
Maybe throw in 2 or 3 fat diodes in series before the dc/dc ?

2bitmarv, 2019-09-17:
Took out the BT module and tested it. The DC/DC converter outputs 4.5V on 14.4V input and ist still working. The BT chip-assembling is dead (tested via 3.3V supply directly on the board at the "OBKVIEVG" connectors). Obviously this BT module has no overload protection and died immediately that moment 14.4V came to it. Maybe another point is a load peak that came across.

To be on the save side i will use a adjustable Down-Step DC Converter before the whole board to ensure that only 12V will be served and the Arduino isn't stressed too.

2bitmarv, 2019-09-17:
> Maybe throw in 2 or 3 fat diodes in series before the dc/dc ?

Mmmmh, would another N4004 after the first one before entering the DC/DC do the job? Since this would drop the current another 0.7 Volts.

archi, 2019-09-17:
_I sent this earlier today via mail, but seems it did not make it here, so here is a revised comment_

I thought the HU provides stabilized 12V on the RUN line? Or are you pulling power from the unstabilized 12V, e.g. cigarette lighter (which is at around 14.4V during driving)?
Did you measure the actual voltage?

Also remember the diode on the board should drop the voltage (not current :P) from 14.4V to 13.7V (or 12V to 11.3V if taken from the HU). Just out of curiosity: Can you measure the DC/DC converter's output voltage with 12V and 13.7V input? I'd suspect yours is broken (or at least on the high end of the spec... +/-20% load regulation is not exactly outstanding).

I fried a BT module when I connected my cable the wrong way around; if I still have it, I'll measure its output voltage with different inputs.

Additional N4004 should do the job. They can take up to 1A, and I think I measured something like 60mA for the whole circuit. So no problem adding a few more. The step-down converter seems like overkill to me, and could cause you problems with noise (the cleaner the input voltage, the better).

Ah, also: These modules are so cheap due to the non-existent QA, maybe you just had bad luck :(

2bitmarv, 2019-09-17:
@archi: Hey Sebastian :), thanks for the good work. I really appreciate your PCB! First i was using the jumpered setting and used to connect to the RUN line. But unfortunately my BT module wasn't responding that well (some less blinking but no connection offering). After that i removed the jumpered setting and used my desk power supply directly. The BT module then worked fine.

Tomorrow i will read the values from the DC/DC and post them here. I also will read the green HU harness connector in my car and simulate that with my desk supply to find a stabilized source inside the HU that fits the drawing for the PCB better than the RUN line

2bitmarv, 2019-09-18:
OK guys, here are the results. First i measured the battery in the trunk and it gives me a constant **12.51V**. The i continued with the BAT+ at the harness that directly goes to the HU which shows a solid **11.85V** while IGN OFF. If ON (POS 1) it slightly drops to **11.60V** and which is now the same at the now enabled ACC connector (the one i preferred for supply the board). Now i tried to drop it trough all the consumers in the car by being switched on/off to create more draw. The only remarkable differences occurred while turning up/down the fan. Putting the fan on MAX the voltage dropped to **11.28V**. Fan on MIN was **11.58V**. While the other consumers like seat warmer, mirror- and backscreen-heating only enabled while engine is running no further drops showed up.

After urning on the engine the voltage rises up to **14.4V**. Putting all available consumers in the climate section to ON and MAX the voltage dropped to **13.75V**.

Back at my desk i supplied the radio with the MAX value of 14.4V and the MIN value of 11.28V to measure the whole supply board of the HU down at the board connectors that goes down in the "main-board". Since i thought there would be a regulated value it shows off that the main power stays almost at the same values. The radio obviously can handle these differences.

So our module drops the Voltage by ~0.7V and therefore the circuit gets a voltage in between 10.65V and 13.75V and the DC/DC on the BT module board get it down to in between 3.02V (IGN at POS 1 while engine off and fan on MAX wich is not really a realistic usecase) and 4.04V. So after all this should be fine since the CS8645 can be supplied by a battery of 3.3V-4.2V.

Putting another N4004 in series before the BT module would drop the supply too much i guess. So the question still is: what blew the BT module off (remember the DC/DC ist still working, it's only the CSR8645 assembling on that board).

archi, 2019-09-18:
Hm, that's difficult to asses from afar ;-) I definitely killed one BT IC by plugging in my cable the wrong way around; maybe your cable is wrong? Did you check the the voltage levels with your cable plugged into the HU (but NOT into the PCB)?

Other than that,... no idea. I designed the board to be used with the 12V from the RUN line, which is stabilized by the HU, so even if your alternator pushes the car's voltage to 14.4V, the PCB only gets 12V (and with the single diode the resulting 11.3V are already a little bit on the low end - e.g. for me the BT sometimes didn't properly restart waiting too long in ON/POS1 before starting the ignition; past tense because the S60 is now gathering rust [still need to sell it] and the "new" V60 D6 has BT built-in).

2bitmarv, 2019-09-19:
Got it! So my BT module blew off by not using the RUN line. I measured the RUN line and in my case it's not really stabilized (in my understanding this means a stabilized source is stepping down or up depending on its input to hold a wanted value). In my V70II the RUN line itself drops the 14.4V (engine running!) by 1V and the N4004 on the board drops it another 0.7V so that the BT modules DC/DC converter drops it to the nearly perfect 3.3V. This clearly explains why the BT module doesn't work when the engine is not running because the supplied voltage is too low at the end for the BT module. I can confirm that you have to go quickly from POS1 to IGN otherwise the BT module won't start correctly. Assumed every RUN line in Volvo behave like that, a solution would be indeed to connect the VCC of the BT module directly to the RUN line before the N4004 but this also means if the engine is running the BT module gets a peak of 4.1V wich is damn near to the max. of 4.2V the module can digest.

2bitmarv, 2019-09-21:
It's pretty annoying...the Voltages are playing merry-go-round. The BT module mostly shows in my display a nearly empty supply (assuming a battery with a voltage between 3.3V and 4.2V i guess) and therefore reminds me with an alert sound from time to time. Id did not expect the circuit of a car to be so unstable like that.