visualapproach/Volvo-melbus #15: Old Files

mikehamm, 2017-11-12:
Hi, this is a fantastic project and it's amazing the work you've done! Apologies for asking this so soon after you've updated the code but would it be possible to get a copy of the old files? 
I'm working on a halfway house between your Bluetooth system and the original 3.5mm audio only system that'll let me change tracks using a 3 ring rca connector and was hoping to borrow some of the code you used to control your footwell lights! Thanks!

mikehamm, 2017-11-12:
Realised after more digging about the code is incorporated into the new files!

closed 2017-11-12 by mikehamm

visualapproach, 2017-11-12:
Hi and thank you! Of course I’ll put them back. I’m happy that someone is interested. I just need some sleep first. (The footwell lights code is still there in the new code but I understand that you wanna continue where you were)

mikehamm, 2017-11-12:
Wonderful thank you so much! Yeah sorry I had some of the files downloaded but for some reason didn't fully clone the repo. Don't rush though. Thanks for all the work you've done on this too!