visualapproach/Volvo-melbus #17: Power on BT module only if HU in CD-changer mode

BK-Poland, 2018-09-03:
Great job! Thank you! I have only one "little" dream: Is it possible to switch power on the BT module only when I switch to the CD changer in HU? BT turned off when I listen to FM radio.

archi, 2018-09-03:
On the hardware side you could add a relay or a suitable NPN transistor to control the power of the module. Software-wise: The protocol allows for this for sure, but @visualapproach would need to add some code to handle that (I didn't look at the code, so no idea how possible that is).

BK-Poland, 2018-09-03:
Hardware for me is no problem. The Arduino output controls the transistor that powers the BT module. The question is: it can be programmed?  @visualapproach ? Does Arduino get a message from HU that the CD-changer is on?

visualapproach, 2018-09-03:
Sure thing, I Will have a look at it and post somerhing soon.

BK-Poland, 2018-09-03:
I will be waiting for a good news. :-)

visualapproach, 2018-09-03:
check lines 43 and 713-740
That should do it, I think

BK-Poland, 2018-09-04:
I have connected according to: older_stuff/schematics_melbus_hack_v2.
All I need to do is change the PLAY PIN from 12 to 10?
And connect Arduino PIN 5 to base of NPN transistor?

visualapproach, 2018-09-04:
No, you shouldn't have to change PLAY PIN. What you wanna do is to capture POWERON-message and switch on the transistor. If POWEROFF is received, you turn off the transistor. That's all you need to change.

If you use the old code (v7i line 255 onwards), look at the following snippets and adjust according to my suggestion:

             //6, power up. resp ack (0x00), not verified
              case 6:
                byteToSend = 0x00;
                SendByteToMelbus();
                trackInfo[1] = startByte;
                trackInfo[8] = startByte;
-->   digitalWrite(5, HIGH); //change "5" to whatever pin you have connected to the transistor.
                break;
              //7, power down. ack (0x00), not verified
              case 7:
                byteToSend = 0x00;
                SendByteToMelbus();
                trackInfo[1] = stopByte;
                trackInfo[8] = stopByte;
-->   digitalWrite(5, LOW); //change "5" to whatever pin you have connected to the transistor
                break;

If you are using the newest code, do the same, but at lines 713 onwards:

            //CDC_PUP
              case 27:
                byteToSend = 0x00;
                SendByteToMelbus();
                trackInfo[1] = startByte;
                trackInfo[8] = startByte;
-->   digitalWrite(5, HIGH); //change "5" to whatever pin you have connected to the transistor.
                break;

              //CDC_PDN
              case 28:
                byteToSend = 0x00;
                SendByteToMelbus();
                trackInfo[1] = stopByte;
                trackInfo[8] = stopByte;
-->   digitalWrite(5, LOW); //change "5" to whatever pin you have connected to the transistor
                break;

Ofc remove the arrows. They are just for your attention.

visualapproach, 2018-09-05:
Yes. Wish you luck, and let us know if it works.

archi, 2018-09-07:
@visualapproach Maybe you can patch this in the repo; I think the feature doesn't do any harm if it goes unused.

BK-Poland, 2018-09-07:
Wait. First, I'll do the tests. Then I will draw a connection diagram. OK? :-)

visualapproach, 2018-09-07:
Yeah sure. If it works for BK-Poland I'll update the repo

visualapproach, 2018-09-07:
And thanks all for keeping this project alive!

BK-Poland, 2018-09-12:
The first test (older stuff code) was quite good. There is only one problem. Output D5 is in a high state also when I listen to the internal CD (HU).
When I switch to FM, BT turns off. Perfect!
When I turn on the CD-Changer, BT turns on. Perfect!
But when I turn on the Internal-HU-CD, BT also turns on. It looks like Internal-HU-CD = CD-Changer.

visualapproach, 2018-09-12:
Ok I’ll have a look at it.

visualapproach, 2018-09-12:
@BK-Poland I couldn't find anything obvious. What happens when you go from CDC->FM->CD ? Does the D5 go high->low->high?

BK-Poland, 2018-09-12:
> What happens when you go from CDC->FM->CD ? Does the D5 go high->low->high?

Yes, exactly.

BK-Poland, 2018-09-12:
There is one more interesting thing. When I switch CDC-> CD, the BT module will restart. So Arduino turns off D5 at 100ms or maybe max 200ms. You can see how the LED in the BT module goes out for a moment.

visualapproach, 2018-09-12:
Can you hook up a computr via USB and log serial output?

visualapproach, 2018-09-12:
(Don’t forget to uncomment the transmission logging in Main loop)

BK-Poland, 2018-09-12:
> Can you hook up a computr via USB and log serial output?

Sure. But... How can I do this? :-)

visualapproach, 2018-09-12:
Arduino in car, hooked up to computer via USB. Use arduino IDE to Reprogram arduino with new software (uncomment the logging part by removing the double slashes). Then turn on radio and watch the serial monitor in IDE. Switch between sources and then copy text

BK-Poland, 2018-09-12:
OK. I need 30 minutes. I'll be back... :-D

BK-Poland, 2018-09-12:
I do not know what I'm doing wrong, but the serial monitor does not work. Communication is correct, but nothing is there. Empty serial monitor window.

visualapproach, 2018-09-12:
Need to sleep now. I get back to you. Check correct port and baudrate 115Kbps (i think)

BK-Poland, 2018-09-13:
> Check correct port and baudrate 115Kbps (i think)

They are correct. I need to learn Arduino IDE :-)

visualapproach, 2018-09-13:
There is no stupid questions. I think this part should be enough:

  //Incoming serial data is supposed to look like this:
  //index, databyte: "3, 5"
  //No error checking here since we're just hacking
//  if (Serial.available() > 0) {
//    int index = Serial.parseInt();
//    trackInfo[index] = (byte) Serial.parseInt();
//    Serial.readStringUntil('\n');
//    for (byte b = 0; b < 9; b++) {
//      Serial.print(trackInfo[b], HEX);
//      Serial.print("-");
//    }
//    Serial.println();
//  }

-->

  //Incoming serial data is supposed to look like this:
  //index, databyte: "3, 5"
  //No error checking here since we're just hacking
 if (Serial.available() > 0) {
    int index = Serial.parseInt();
    trackInfo[index] = (byte) Serial.parseInt();
    Serial.readStringUntil('\n');
    for (byte b = 0; b < 9; b++) {
      Serial.print(trackInfo[b], HEX);
      Serial.print("-");
   }
   Serial.println();
}

visualapproach, 2018-09-13:
@BK-Poland 
Good, you got serial monitoring to work!

Here is a "translation" of the log:

`
(connect to Arduino, HU is on FM)
7 1A EE F8 7F
E8 1E EF (Cartridge info request)

E9 1B E0 1 8 track info request)
(FM to CDC) -> E8 19 2F (****CDC Power UP****)
E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)
E8 19 22 (**CDC Power DOWN**)
81 1B E0 1 8 10 2 1 1 80 1 C CC CC
E9 1B E0 1 8 track info request)
80 19 2F 0 E 76 43 C3 FC C0 70 A4 0 80 C0 0 87 FF FF
(CDC to CD) -> 81 1B E0 1 8 10 48 1 1 80 1 F FF FF

81 1B E0 1 8 10 48 1 1 80 1 F FF FF

81 1B E0 1 8 10 48 1 1 80 1 F FF FF

81 1B E0 1 8 10 48 1 1 80 1 F FF FF

81 1B E0 1 8 10 48 1 1 80 1 F FF FF

81 1B E0 1 8 10 48 1 1 80 1 F FF FF
0 1C EC

81 1B E0 1 8 10 8 1 1 80 1 0 5 32
0 1C EC
81 1B E0 1 8 10 8 1 1 80 1 0 5 33

81 1B E0 1 8 10 8 1 1 80 1 0 5 33
0 1C EC
81 1B E0 1 8 10 8 1 1 80 1 0 5 34

81 1B E0 1 8 10 8 1 1 80 1 0 5 34
0 1C EC
81 1B E0 1 8 10 8 1 1 80 1 0 5 35

81 1B E0 1 8 10 8 1 1 80 1 0 5 35
0 1C EC
81 1B E0 1 8 10 8 1 1 80 1 0 5 36

81 1B E0 1 8 10 8 1 1 80 1 0 5 36
0 1C EC
81 1B E0 1 8 10 8 1 1 80 1 0 5 37
80 19 22 0 E 76 43 C3 FC C0 70 A1 0 80 C0 0 86 66 66
21:10:46.682 -> E9 1B E0 1 8 track info request)
81 1B E0 1 8 10 42 1 1 80 1 C CC CC
E8 19 2F (****CDC Power UP****)
(CD to CDC) -> E9 1B E0 1 8 track info request)
0 1C EC
E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)

E9 1B E0 1 8 track info request)
E8 19 22 (**CDC Power DOWN**)
(CDC to FM) -> E9 1B E0 1 8 track info request)

80 19 22 0 E 76 43 C3 FC C0 70 81 0 C0 C0 0 86 66 66
(Shut down HU) -> E8 19 22 (**CDC Power DOWN**)
81 1B E0 1 8 10 2 1 81 80 1 C CC CC
21:11:04.113 -> E9 1B E0 1 8 track info request)

0 18 12
`

It appears the communication is good and the HU does the right thing. That narrows it down to program error or hardware error.
Please add following lines to your code and reupload, then check serial output again.

after case 6, directly after the line "digitalWrite(5, HIGH);" add this

Serial.println("***D5 ON***");

after case 7, directly after the line "digitalWrite(5, LOW);" add this

Serial.println("***D5 OFF***");

visualapproach, 2018-09-13:
You are also welcome to post that part of your code if you want me to check for errors.

BK-Poland, 2018-09-14:
7 1A EE F8 7F 
E8 1E EF 
E9 1B E0 1 8 
D5 ON
E8 19 2F 
E9 1B E0 1 8 
E9 1B E0 1 8 
E9 1B E0 1 8 
E9 1B E0 1 8 
E9 1B E0 1 8 
E9 1B E0 1 8 
E9 1B E0 1 8 
E9 1B E0 1 8 
E9 1B E0 1 8 
D5 OFF
E8 19 22 
81 1B E0 1 8 10 2 1 1 80 1 C CC CC 
E9 1B E0 1 8 
D5 ON
80 19 2F 0 E 76 43 C3 FC C0 70 A4 0 80 C0 0 80 2 A0 
81 1B E0 1 8 10 48 1 1 80 1 0 5 40 
81 1B E0 1 8 10 48 1 1 80 1 0 5 40 
81 1B E0 1 8 10 48 1 1 80 1 0 5 40 
81 1B E0 1 8 10 48 1 1 80 1 0 5 40 
81 1B E0 1 8 10 48 1 1 80 1 0 5 40 
81 1B E0 1 8 10 8 1 1 80 1 0 5 40 
0 1C EC 
81 1B E0 1 8 10 8 1 1 80 1 0 5 40 
0 1C EC 
81 1B E0 1 8 10 8 1 1 80 1 0 5 41 
D5 OFF
80 19 22 0 E 76 43 C3 FC C0 70 A1 0 80 C0 0 86 66 66 
E9 1B E0 1 8 
81 1B E0 1 8 10 42 1 1 80 1 C CC CC 
D5 ON
E8 19 2F 
E9 1B E0 1 8 
0 1C EC 
E9 1B E0 1 8 
E9 1B E0 1 8 
E9 1B E0 1 8 
E9 1B E0 1 8 
E9 1B E0 1 8 
D5 OFF
E8 19 22 
E9 1B E0 1 8 
D5 OFF
80 19 22 0 C 91 0 
81 1B E0 1 8 10 2 1 1 80 1 C CC CC 
E9 1B E0 1 8

visualapproach, 2018-09-14:
yeah it's code related. I'll investigate. Could you send me your code by email?

BK-Poland, 2018-09-14:
/*
  new: 
  Press cd1 or cd2 to change bluetooth volume.
  Cd change is resetting track number.

  
  By Thomas Landahl, 2017-04-25

*/

#define MELBUS_CLOCKBIT (byte)2 //Pin D2  - CLK
#define MELBUS_DATA (byte)3     //Pin D3  - Data
#define MELBUS_BUSY (byte)4     //Pin D4  - Busy
const byte prevPin = 8;
const byte nextPin = 9;
const byte upPin = 10;    //volume up
const byte downPin = 11;  //volume down
const byte playPin = 12;

//volatile variables used inside and outside of ISP
volatile byte melbus_ReceivedByte = 0;
volatile byte melbus_Bitposition = 7;
volatile bool byteIsRead = false;

byte byteToSend = 0;  //global to avoid unnecessary overhead
unsigned long Connected = 0;

//preset parameters
byte track = 0x01; //Display show HEX value, not DEC. (A-F not "allowed")
byte cd = 0x01; //1-10 is allowed (in HEX. 0A-0F and 1A-1F is not allowed)
byte powerup_ack = 0x00;

//Base adresses.
//const byte MD_ADDR = 0x70;  //internal
//const byte CD_ADDR = 0x80;  //internal
//const byte TV_ADDR = 0xA8;  //might be A9 during main init sequence
//const byte DAB_ADDR = 0xB8;
//const byte SAT_ADDR = 0xC0;
//const byte MDC_ADDR = 0xD8;
//const byte CDC_ADDR = 0xE8;
//const byte RESPONSE = 0x06; //add this to base adress when responding to HU
//const byte MASTER = 0x07;   //add this to base adress when requesting/beeing master

//change theese definitions if you wanna emulate another device.
//My HU-650 don't accept anything but a CD-C (so far).
#define RESPONSE_ID 0xEE  //ID while responding to init requests (which will use base_id)
#define MASTER_ID 0xEF    //ID while requesting/beeing master
#define BASE_ID 0xE8      //ID when getting commands from HU
#define ALT_ID 0xE9       //Alternative ID when getting commands from HU

//This list can't be too long. We only have so much time between the received bytes. (approx 500 us)
const byte commands[][5] = {
  {BASE_ID, 0x1E, 0xEF},             //0, Cartridge info request. Respond with 6 bytes (confirmed)
  {ALT_ID, 0x1B, 0xE0, 0x01, 0x08},  //1, track info req. resp 9 bytes
  {BASE_ID, 0x1B, 0x2D, 0x40, 0x01}, //2, next track.
  {BASE_ID, 0x1B, 0x2D, 0x00, 0x01}, //3, prev track
  {BASE_ID, 0x1A, 0x50},             //4, change cd
  {BASE_ID, 0x1A, 0x50},             //5, not used
  {BASE_ID, 0x19, 0x2F},             //6, power up. resp ack (0x00), 0x19 could be 0x49??
  {BASE_ID, 0x19, 0x22},             //7, power down. ack (0x00), not verified
  {BASE_ID, 0x19, 0x29},             //8, FFW. ack, not verified
  {BASE_ID, 0x19, 0x26},             //9, FRW. ack, not verified
  {BASE_ID, 0x19, 0x2E},             //10 scan mode. ack, cmd verified!
  {BASE_ID, 0x19, 0x52},             //11 random mode. ack, not verified
  {0x07, 0x1A, 0xEE},                //12 main init seq. wait for BASE_ID and respond with RESPONSE_ID.
  {0x00, 0x00, 0x1C, 0xED},          //13 secondary init req. wait for BASE_ID and respond with RESPONSE_ID.
  {0x00, 0x1C, 0xEC}                 //14 master req broadcast. wait for MASTER_ID and respond with MASTER_ID.
};
//keep track of the length of each command. (The two dimensional array above have fixed width (padded with 0x00))
const byte listLen = 15;
[... 506 lines cut ...]

void play() {
  digitalWrite(playPin, HIGH);
  for (byte i = 0; i < 200; i++)
    delayMicroseconds(1000);
  digitalWrite(playPin, LOW);
}

void volumeDown() {
  digitalWrite(downPin, HIGH);
  for (byte i = 0; i < 200; i++)
    delayMicroseconds(1000);
  digitalWrite(downPin, LOW);

}

void volumeUp() {
  digitalWrite(upPin, HIGH);
  for (byte i = 0; i < 200; i++)
    delayMicroseconds(1000);
  digitalWrite(upPin, LOW);

}

//Happy listening, hacker!

visualapproach, 2018-09-14:
if ((matching[cmd] == cmdLen[cmd]) && (byteCounter+1 ==cmdLen[cmd] )) {

BK-Poland, 2018-09-15:
It works! Perfectly! A big beer for you. Thank you!

BK-Poland, 2018-09-15:
OK. Time to sharezone my hardware. I use  the MOSFET IRL540N (logic level). I use gate resistor 1Mohm. Its simple to make a new function :-)

visualapproach, 2018-09-15:
Awesome! Cheers

BK-Poland, 2018-09-17:
Do you need me to draw my wiring diagram? If I can help, please write. You can publish my idea in the official code.

BK-Poland, 2018-09-17:
Why do I use the old code? It's simpler and it's enough for me. It's easier for me to analyze it. It meets all my requirements. BT works in my old Volvo. Thank you very much for that! When I was doing my BT-Volvo-Car according to your instructions and with your code it was still an old code. Maybe I will try a new code in the future. In what program do you draw your electrical diagrams?

visualapproach, 2018-09-18:
Ok, I understand. If you are happy then no need to switch code :-)
I've used Eagle. Supposedly you can import your PCBs into Fusion360 wich is an advantage. But it is a rather complicated program for what I do. I have also peeked at easyEDA, an online PCB editor.

BK-Poland, 2018-10-04:
I remember that I had to draw a diagram. I'm working on one more improvement. I am waiting for a shipment  (with some electronics parts) from China. A month or two. Please be patient. :-)

visualapproach, 2018-10-04:
No worries. I will update the code also. Some day :-)

BK-Poland, 2018-10-08:
I have a problem with the hardware layer. Sometimes the BT module turns itself on. Sometimes Arduino goes crazy and I have CD Error on the HU display. Even, he sometimes switches the songs himself. It seems to me that disconnecting the mass of the BT module is not a good solution. If I do a properly working layout, I will show the diagram. Please be patient. :-D

BK-Poland, 2019-02-03:
I found the fault. The Arduino Nano stabilizer on the PCB was damaged. The damage was revealed only after a few minutes of work. But sometimes only after an hour. I used a step-down converter and Arduino has 5V power supply.

I have one more request for help: Is it possible to do so that Arduino does not turn off BT when shooting the starter?

BK-Poland, 2019-02-03:
Yes I know. I've already done direct battery power. Arduino and BT have power supply always.
When I listen music in parking, for example Spotify and when I start the engine, it disconnects the phone and BT, and then I have to wait until the phone connects to BT and then I need to press PLAY. I would like to avoid it. Will you help me? :-)

visualapproach, 2019-02-22:
any success?

BK-Poland, 2019-02-26:
I am sorry that I have not written for so long. I had a lot of work and not enough free time.

Unfortunately it does not work. Normally turns BT on and off without delay. Where should I insert "if (mytimer_enabled) ...."? Are you sure that 5 seconds is 500000? If in milliseconds it should probably be 5000?

BK-Poland, 2019-02-26:
/*
  new: 
  Press cd1 or cd2 to change bluetooth volume.
  Cd change is resetting track number.

  
  By Thomas Landahl, 2017-04-25

*/

#define MELBUS_CLOCKBIT (byte)2 //Pin D2  - CLK
#define MELBUS_DATA (byte)3     //Pin D3  - Data
#define MELBUS_BUSY (byte)4     //Pin D4  - Busy
const byte prevPin = 9;
const byte nextPin = 8;
const byte upPin = 10;    //volume up
const byte downPin = 11;  //volume down
const byte playPin = 12;

//volatile variables used inside and outside of ISP
volatile byte melbus_ReceivedByte = 0;
volatile byte melbus_Bitposition = 7;
volatile bool byteIsRead = false;

byte byteToSend = 0;  //global to avoid unnecessary overhead
unsigned long Connected = 0;

//preset parameters
byte track = 0x01; //Display show HEX value, not DEC. (A-F not "allowed")
byte cd = 0x01; //1-10 is allowed (in HEX. 0A-0F and 1A-1F is not allowed)
byte powerup_ack = 0x00;

//Base adresses.
//const byte MD_ADDR = 0x70;  //internal
//const byte CD_ADDR = 0x80;  //internal
//const byte TV_ADDR = 0xA8;  //might be A9 during main init sequence
//const byte DAB_ADDR = 0xB8;
//const byte SAT_ADDR = 0xC0;
//const byte MDC_ADDR = 0xD8;
//const byte CDC_ADDR = 0xE8;
//const byte RESPONSE = 0x06; //add this to base adress when responding to HU
//const byte MASTER = 0x07;   //add this to base adress when requesting/beeing master

//change theese definitions if you wanna emulate another device.
//My HU-650 don't accept anything but a CD-C (so far).
#define RESPONSE_ID 0xEE  //ID while responding to init requests (which will use base_id)
#define MASTER_ID 0xEF    //ID while requesting/beeing master
#define BASE_ID 0xE8      //ID when getting commands from HU
#define ALT_ID 0xE9       //Alternative ID when getting commands from HU

//This list can't be too long. We only have so much time between the received bytes. (approx 500 us)
const byte commands[][5] = {
  {BASE_ID, 0x1E, 0xEF},             //0, Cartridge info request. Respond with 6 bytes (confirmed)
  {ALT_ID, 0x1B, 0xE0, 0x01, 0x08},  //1, track info req. resp 9 bytes
  {BASE_ID, 0x1B, 0x2D, 0x40, 0x01}, //2, next track.
  {BASE_ID, 0x1B, 0x2D, 0x00, 0x01}, //3, prev track
  {BASE_ID, 0x1A, 0x50},             //4, change cd
  {BASE_ID, 0x1A, 0x50},             //5, not used
  {BASE_ID, 0x19, 0x2F},             //6, power up. resp ack (0x00), 0x19 could be 0x49??
  {BASE_ID, 0x19, 0x22},             //7, power down. ack (0x00), not verified
  {BASE_ID, 0x19, 0x29},             //8, FFW. ack, not verified
  {BASE_ID, 0x19, 0x26},             //9, FRW. ack, not verified
  {BASE_ID, 0x19, 0x2E},             //10 scan mode. ack, cmd verified!
  {BASE_ID, 0x19, 0x52},             //11 random mode. ack, not verified
  {0x07, 0x1A, 0xEE},                //12 main init seq. wait for BASE_ID and respond with RESPONSE_ID.
  {0x00, 0x00, 0x1C, 0xED},          //13 secondary init req. wait for BASE_ID and respond with RESPONSE_ID.
  {0x00, 0x1C, 0xEC}                 //14 master req broadcast. wait for MASTER_ID and respond with MASTER_ID.
};
//keep track of the length of each command. (The two dimensional array above have fixed width (padded with 0x00))
const byte listLen = 15;
[... 524 lines cut ...]
    delayMicroseconds(1000);
  digitalWrite(playPin, LOW);
}

void volumeDown() {
  digitalWrite(downPin, HIGH);
  for (byte i = 0; i < 200; i++)
    delayMicroseconds(1000);
  digitalWrite(downPin, LOW);

}

void volumeUp() {
  digitalWrite(upPin, HIGH);
  for (byte i = 0; i < 200; i++)
    delayMicroseconds(1000);
  digitalWrite(upPin, LOW);

}

void shutoffBT() {
digitalWrite(5, LOW);
Serial.println("D5 OFF");
}
//Happy listening, hacker!

visualapproach, 2019-02-26:
This line: static long unsigned int mytimer = 0;
should be static unsigned long mytimer = 0; 
(my bad)
You have mixed D5 and D6, and HIGH/LOW logic. Assuming D6 is controlling your mosfet, and I will write ON/OFF instead of HIGH/LOW. Depending on your circuit you need to replace with the proper one.
Case 6: change D6 to ON (HIGH or LOW)
Case 7: remove the line "digitalWrite(6...". You wanna wait before doing that.
Lastly: in the shutoffBT() -  D6 instead of D5 and HIGH/LOW should be OFF

> Are you sure that 5 seconds is 500000? If in milliseconds it should probably be 5000?

I'm sure it is not 5 s. It's not a timer per se, it's a counter. It would be alot easier to time things if timers where on, but they might interfere with communication. As I commented in the code, you need to adjust the value yourself. If it takes 20 s to count to 500000 then you try 125000.

BK-Poland, 2019-02-28:
It works! Exactly as I wanted. Thank you! :-)
I changed D5 to D6 because I probably damaged D5. I also changed the high / low logic, because now I control the relay module. No MOSFET.
Thank you very much!

visualapproach, 2019-03-01:
You’re welcome

BK-Poland, 2019-03-01:
What kind of program did you use to draw the scheme of your project?

visualapproach, 2019-03-01:
Eagle

hagurel, 2019-03-15:
hepinizi tebrik ederim

visualapproach, 2019-03-15:
Tack ska du ha