visualapproach/Volvo-melbus #19: Ticking noise

RSchalkwijk, 2018-10-01:
Hi thanks for the good documentation / schematics / descriptions. 
I've implemented the project with some success however the constant ticking gremlin is really keeping me from claiming this project successful. 
My first thought was that it was a grounding issue however I've tried every conceivable combination of power and grounding options with no success. 
I believe there could be an issue with the code as the ticking is dependent on the skip button as well. 
On my project the ticking is constant for about 50 seconds then briefly goes away.  
I've hooked up a logic analyzer to the data,busy and clock lines while connected to the radio. 
Could I send this to you to have a look at and confirm my suspicion that the issue is code related?

visualapproach, 2018-10-01:
Hi @RSchalkwijk! Welcome and thanks!
Assumption: you are using the code in this repository.
The code is responsible for sending data over Melbus lines. That in turn makes electronical noise which is picked up by the audio lines. So the only way to make silence BY CODE is to stop sending data, which contradicts the purpose of the device. Solution is in the hardware. 
What bluetooth module do you have? 
Are you using a Ground Loop Isolator? 
I can recommend the CSR8645 BT module. I think @archi is using it without a GLI. The old DIY module is notoriously difficult to get noise free and definately needs one.

RSchalkwijk, 2018-10-01:
I've tried a ground loop isolator which did not help. At the moment I have no audio lines connected to the audio R,L,Common. I tried to simplify everything in order to isolate possible noise. I used an original volvo din cable that plugs into the radio and a female socket - the socket connects to my proto-board.
I've used heat shrink and insulation to ensure that there is no connection form the melbus to the audio lines. The weird thing is that the noise disappears every 50ish seconds for about 4 seconds... 

If messed around a bit a noticed that if  you disconnect the device from the radio for some reason it sometimes still registers the CD-C... The ticking sound is then more regular with no break in the repeating pattern - so where does that noise come from?  Was thinking its a possible noise that comes if the request for reply from the HU is NAC'd?

RSchalkwijk, 2018-10-01:
I can mail you the logs and traces from the logic analyzer? My coding skills are very limited, would be great if you can verify what the ticking is -  I suspect others have  possibly picked up the same issue. The ticking sound feels like some sort of barbaric torture to me. Was thinking of buying an aftermarket product (Yatour,grom...) but not sure if it will fix the issue if its something to do with my HU. Thanx again!

visualapproach, 2018-10-01:
Yes the HU is keeping the input open but will eventually show [source] error. Even if you unplug the cable, the HU keeps sending data on Melbus lines while talking to the internal CD player. Thats most likely what's get picked up by the audio lines, internal to the HU. So while plugging the arduino in might make the noise louder, since the cables might act as an antenna, it is no longer the source of the "data noise". I also had the ticking problem, but changing BT-module to CSR solved that. Me and others are using the same code without problem, so it is not code related (IMHO ;), so analyzing the data is not meaningful i think. But if it makes you feel better, sure. Send it and I'll check when I got some spare time.

RSchalkwijk, 2018-10-01:
Cool thanx, I'll try that BT module first. One question does the CSR you use have a "pre-amp"? The ticking noise is only really audible from say 25% HU volume but that is way too soft if the audio lines are being used for an aux cable or for the BT module I tried

archi, 2018-10-02:
Hey, remote diagnostics are difficult, but you're saying you didn't connect the audio lives to anything? My first try would be to terminate those, since they're going to pick up a lot of noise in a car. Just short L and R to audio common ground using a 4k7 resistor each and see if that helps.

Imho that's the most likely culprit. Back when I was using an aux cable instead of a BT module I had a lot of noise while the cable was unplugged and the radio set to CD-Changer/Aux.

archi, 2018-10-02:
Also, yes, the module has a pre-amp. They claim to use a specialized TI headphones preamp (dual op amp plus some other features; much better than general purpose parts). The part is quite small, so I couldn't check the markings, but I believe they've actually used it.

Bonus: The module even has an isolated audio ground, which is why *that* module doesn't need a ground loop breaker :)

BK-Poland, 2018-10-04:
I have the same problem. 560 ohm resistors help but not quite. Currently, I have the audio cable and the signal cable together on the length of the meter. I plan to shorten and separate audio cable. Maybe I'll even put on a shielded audio cable. What do you think, visualapproach?

visualapproach, 2018-10-04:
I think a CSR8645 is less hassle and better result. Else a GLI is a must and i am not sure how well it Will work.

AudioDrugs, 2018-11-13:
Interesting. Im using original cd-changer cable. Actually no audio connected. Noise from data pins. Any solution?

Video:
[20181113_224247_1.zip](https://github.com/visualapproach/Volvo-melbus/files/2578046/20181113_224247_1.zip)

BK-Poland, 2018-11-14:
What is a "GLI" ? :-)

Arroquw, 2020-04-08:
> Hey, remote diagnostics are difficult, but you're saying you didn't connect the audio lines to anything? My first try would be to terminate those, since they're going to pick up a lot of noise in a car. Just short L and R to audio common ground using a 4k7 resistor each and see if that helps.
> 
> Imho that's the most likely culprit. Back when I was using an aux cable instead of a BT module I had a lot of noise while the cable was unplugged and the radio set to CD-Changer/Aux.

This is what happening to me now, adding the resistors doesn't help at all though. When not even my board is connected and there is just the cable (after having it initialise as CDC) the ticking noise will still continue. GLI doesn't work either

AudioDrugs, 2024-03-02:
Hey guys.
Did some one managed how to reduce the noise?

I made a quite good 4 layer PCB with FSC-1026D (QCCC3034) and DAC and found that all these fancy things are not helping. In this video music was paused and HU volume was set to max. HU constantly sending a master and track info requests. Is it possible to prevent HU from requesting track info? The same noise remains while running module is unplugged.

https://github.com/visualapproach/Volvo-melbus/assets/29013110/b4f29dac-7d09-4941-bbc1-d48249ff6ff2

Aslo I have a BlueMusic emulator which is based on PIC16F and has 100ohm termination, but in this module data noise is ultra low. And it remains low even running module is unplugged. I guess it is possible to stop HU from sending these requests.