visualapproach/Volvo-melbus #24: Doesn't seem to work properly with HU-603

Maxima666, 2019-08-12:
So I've soldered the module as the schedule shown (I guess it's the same as for archi's PCB?) and is using the .ino meant for archi's PCB!
But the text does not come through, and I can't cycle through different texts, it just says "Track 01" and so on, or what ever it's supposed to say with an actual cd changer installed.

Plus, when i press the RND-button, it just works as a Play/Pause-button, odd?

Otherwise it works perfecly!
Unbelievable in audio quality from those cassette-to-3,5mm I've been using before!

best,
Maxima

visualapproach, 2019-08-12:
Hi Maxima, Nice! Yeah, text will only work in SAT mode (not CD Changer mode). Same for RND-button. I don’t know which HU models support SAT, but if yours doesn’t its nothing wrong with your build. If there is a SOURCE selector try to cycle through and look for SAT.

jurgislipsnis, 2019-10-28:
To convert this to work as MD changer, I only need to change BASE_ID? As I have HU803, SAT is not an option for me

visualapproach, 2019-10-28:
@jurgislipsnis It might open the md as audio source but text will not work by just changing base id.

NeoRanga, 2020-01-02:
Hi, I have an HU-603 and I use archi's PCB.
But i can't get it to work properly, when I connect it to the HU I can select the CD-Changer and i hear the BT-module power up thru the speakers and then I think it's powering down and the light on the BT-Module turns off.
If I connect 12v directly to the PCB without the HU the BT-Module stays on and I can connect to it.

visualapproach, 2020-01-02:
Hi, Sounds like a power supply issue then. I'd check voltage at the PCB in car, everything connected and running. Voltage drops may occur due to poor power/GND cord or you may have a short somewhere.
Where do you get 12V from? The DIN-connector's 12V line was weak on my HU so I had to use 12V from the cigarette lighter.
It may also be over voltage. Someone here mentioned that he had that problem IIRC.

NeoRanga, 2020-01-02:
I use the power from the DIN-connector and I will try to use power from cig-lighter instead. But when I tested it on the bench, it draws less then 100mA @ 12v. The HU-603 doesn't have a play button, can I program an alternative button, or is it already done? There isn't some shutdown command that is turning the BT-Module off? cuz the nano appears to keep running.

visualapproach, 2020-01-02:
Try that. There is no BT OFF command by default. Yes you can change the code so any button can do any of the available functions. Read through the code and it should be fairly obvious where to change the button commands. Hope it works out

NeoRanga, 2020-01-05:
Connected it directly to 12v and I could pair the phone and play some music. But after about 1min, the BT-module stopped working. CD-Changer Emu is still working, but I think the BT-module fried.

visualapproach, 2020-01-06:
@NeoRanga check this out
https://github.com/visualapproach/Volvo-melbus/issues/10#issuecomment-460001748

NeoRanga, 2020-01-09:
Thank for the link, but I think something has happened to the bt-module. There is no power light on the module and i have tried to feed it with 14v from a bench power supply, but still nothing.

2bitmarv, 2020-01-19:
@NeoRanga: I think you fried your BT module. So i did and therefore started to inspect some things: #14 
So i invite you to read about it and find out that your car is NEVER serving 12V! Values rise and fall like a rollercoaster depending on how many consumers are working. If you are using the same BT module as me an Archi then things can get annoying because providing this nasty little thang has to be really delicate due to its limited range. I recommend using the RUN line from CD-Changer as it is somehow stabilized. Cig-Lighter Power is much to much. Although the Nano PCB will take it without hassle but your BT module will provided with too much power.

NeoRanga, 2020-01-19:
@2bitmarv Hi, I did use the run line at first, but the BT-module powered up and then directly down. I think the BT-module I used had a 12v regulator, but i have to check that. The module worked for a while when connected to 12v cig. No magic smoke, it just powered down forever. The car wasn't running, so the voltage was around 12,5v.

2bitmarv, 2020-01-19:
@NeoRanga : Ok, another thing might be the various quality of these "cheap" BT modules. These "little things to experiment with" mostly rely on a stabilized power supply. I fried many modules of various types or received them already dead on delivery. Good Luck!

Arroquw, 2020-02-14:
When using the RUN from the DIN instead of 12v from cig lighter, are they just interchangeable in the PCB design? So instead of external 12v line you just connect run to the transistor?

visualapproach, 2020-02-14:
Yes you can use either one. Run line may be regulated but didn’t provade enough current all the time.

VincentGijsen, 2021-06-05:
push https://github.com/VincentGijsen/melbus-sniffed-traffic/commit/59f7c61e5ae8cf85db9b7f11014d31fe805093a9  has 803 hu command logs (MDchanger)