visualapproach/Volvo-melbus #25: suggestion BT-module-upgrade

VincentGijsen, 2019-11-03:
Hi,

Long time. came accros the Microchip module (BM20), and this guy's repo: [https://github.com/tomaskovacik/arduino ](https://github.com/tomaskovacik/arduino) 

These modules also support meta-data and are rather cheap (EURO 6,90 before tax-shipping) and can be purchased single pieces at microchipdirect.com

@tomaskovacik also sells breakout-boards to make it more comfortable to work with the small BT-module: [https://www.tindie.com/products/tomaskovacik/bm20-breadboard-adapter/](https://www.tindie.com/products/tomaskovacik/bm20-breadboard-adapter/)

I just placed order for BM20 and the breakoutboard (so no hands-on experience yet)

disclaimer: i'm not affiliated in any way, just like the guys efforts and open-sourcing 

everything(hardware + software).

i'm thinking about moving to a STM32103 (aka 'blue-pill') and use freertos. than i might implement lcd-scrolling in software (as i havent encountered the particular melbus-flag)

cheers,
Vincent

tomaskovacik, 2019-11-04:
actual library is here: https://github.com/tomaskovacik/IS2020
avrcp is not complete ..

visualapproach, 2019-11-04:
Cool!

VincentGijsen, 2020-01-26:
Hi,
small update
the stated module supports among others AVRCP1.5 and raw-AVRCP commands. with that, i'm able to extract all kind of nifty details.

currently working:
- song-meta info (album/artist/title/length), extracted
- playback-position, 
- player-control (spotify)
- event handling 
- various details like phone-battery, capacity, 
- signal strength, roaming status

likely to get working:
- do navigation of song-repository
- cover art (not so usefull on our volvo radio's)
- phone-book /contact support
 various other stuff 

It uses a uart interface, so hardware-wise not so messy as binary contacts, however software wise quite a bit more complex, with keeping states.

my current software is here https://github.com/VincentGijsen/MelbusRtos should you be interested, however still very rough, no melbus-code implemented yet but as we know how that must work, it just work-ahead, now i focused primarily on getting al bluetooth functions available, and a practical means to test the UI, which i simulate using the usb-stack with vcp on the STM32f bluepill. i'm using FreeRTOS to keep all the tasks /housekeeping separated.

tomaskovacik, 2020-03-23:
guys, I just tested BM62 module which compatible with BM20 (pins and serial protocol), it is flash-based so future firmware upgrades are possible (BM20 is ROM) 

there are also some sources  for pic micros in C, for inspiration 
https://www.microchip.com/wwwproducts/en/BM62