visualapproach/Volvo-melbus #26: Creeping line on the screen HU

assa62, 2019-12-18:
Thanks to the author for the wonderful code with which we can connect BT to our devices. I came across BT that displays track data in the serial port. Here is a little finalized. https://youtu.be/w6zfiRrXcH0
Thanks!

visualapproach, 2019-12-18:
Awesome! Thanks for the kind words. Do you do the scrolling by updating the text very often or did you find a nicer way?

assa62, 2019-12-18:
I can’t defeat the melbus, but I can insert such a piece into your code
`
void setCustomText() {
  byte tmp;
  tmp = customText[textRow - 1][0];
  for (int i = 0; i < 36 - 1; i++)
    customText[textRow - 1][i] = customText[textRow - 1][i + 1];
  customText[textRow - 1][36 - 1] = tmp;
}`

visualapproach, 2019-12-18:
Great, good work!

Murat8684, 2020-01-18:
Hi
aforementioned how do i add code Can you help.

assa62, 2020-01-19:
> Great, good work!

Hi. Can I post your code here with my improvements?

assa62, 2020-01-19:
> Hi
> aforementioned how do i add code Can you help.

Hi. In addition to the code, you must have a shield bluetooth with Rx/Tx

visualapproach, 2020-01-19:
> > Great, good work!
> 
> Hi. Can I post your code here with my improvements?

Yes

assa62, 2020-01-19:
``/*
  New in this version:
    Hardwarewise - Adjusted pin numbers to match archi's PCB v2.0
    Softwarewise - Changed the way RGB LEDs are managed. Press "3" on HU to cycle through colors on driver side.
                    press "4" to do the same on passenger side. See github for more info.
                   Changed customText variable to hold four lines of text.
                   Battery voltage now only show one decimal. It's an estimate anyway.
                   Steering wheel "next/prev" is now working. (Same as pressing 2/6)
                   Switching to/from SAT source toggles a pin (A6). Can be used to turn of devices like the BT module. (github request)
                   (Not tested by me yet)

  Todo (someone): if possible, enable scrolling text. This should be possible since the text messages is larger than the screen.
  I guess scrolling can be turned on by
  1) editing some bits in the init messages
  2) editing some bits in the text headers
  3) some specific character combination in the text itself
  (Text IS scrolling twice after pressing source/sat display/ and choosing title/artist etc on the HU)

  By Thomas Landahl, 2018-10-16
  https://github.com/visualapproach/Volvo-melbus

  Comm sniffing and some code contribution by Vincent Gijsen. Thanks a LOT!

*/

#include <EEPROM.h>

#define INT_NUM (byte)1         //Interrupt number (0/1 on ATMega 328P)
#define MELBUS_CLOCKBIT (byte)3 //Pin D2  - CLK
#define MELBUS_DATA (byte)4     //Pin D3  - Data
#define MELBUS_BUSY (byte)5     //Pin D4  - Busy

/*const byte mutePin = 9;
  const byte playPin = 10;
  const byte prevPin = 11;
  const byte nextPin = 12;
  const byte upPin = 7;    //volume up (not used on PCB)
  const byte downPin = 8;  //volume down (not used on PCB)

  const byte LED_RED_LFT = 14;  //A0
  const byte LED_GRN_LFT = 15;  //A1
  const byte LED_BLU_LFT = 16;  //A2
  const byte LED_RED_RGT = 19;  //A3
  const byte LED_GRN_RGT = 18;  //A4
  const byte LED_BLU_RGT = 17;  //A5
*/
const byte MISC = 20;         //A6
const byte BATTERY = 21;      //A7 - PCB uses this
/*
  byte lftClr;
  byte rgtClr;
  String colors[8] = {
  "Black ", "Blue  ", "Green ", "Cyan  ", "Red   ", "Purple", "Yellow", "White "
  };
*/
const int R_CAR = 10; //cable resistance from battery to device
const int R_1 = 3300; //r1 of voltage divider (connected to GND)
const int R_2 = 6800; //r2 of voltage divider (connected to RAW)
const float T_LOSS = 0.8; //if circuit is protected with a transistor and diode, put total voltage drop here.

byte track = 0x01; //Display show HEX value, not DEC. (A-F not "allowed")
byte cd = 0x01; //1-10 is allowed (in HEX. 0A-0F and 1A-1F is not allowed)

//volatile variables used inside AND outside of ISP
volatile byte melbus_ReceivedByte = 0;
volatile byte melbus_Bitposition = 7;
volatile bool byteIsRead = false;

byte byteToSend = 0;          //global to avoid unnecessary overhead
bool reqMasterFlag = false;   //set this to request master mode (and sendtext) at a proper time.
[... 1274 lines cut ...]
}

void SendTrackInfo() {
  for (byte i = 0; i < 9; i++) {
    byteToSend = trackInfo[i];
    SendByteToMelbus();
  }
}

void SendCartridgeInfo() {
  for (byte i = 0; i < 6; i++) {
    byteToSend = cartridgeInfo[i];
    SendByteToMelbus();
  }
}

float getBatV() {
  float a, bv;

  a = analogRead(BATTERY) * 5.0 / 1023.0;
  bv = a * (R_1 + R_2 + R_CAR) / R_1 + T_LOSS;
  return bv;
}

//Happy listening AND READING, hacker!

assa62, 2020-01-19:
Here is the text for my "tandem"
`/*
  New in this version:
    Hardwarewise - Adjusted pin numbers to match archi's PCB v2.0
    Softwarewise - Changed the way RGB LEDs are managed. Press "3" on HU to cycle through colors on driver side.
                    press "4" to do the same on passenger side. See github for more info.
                   Changed customText variable to hold four lines of text.
                   Battery voltage now only show one decimal. It's an estimate anyway.
                   Steering wheel "next/prev" is now working. (Same as pressing 2/6)
                   Switching to/from SAT source toggles a pin (A6). Can be used to turn of devices like the BT module. (github request)
                   (Not tested by me yet)

  Todo (someone): if possible, enable scrolling text. This should be possible since the text messages is larger than the screen.
  I guess scrolling can be turned on by
  1) editing some bits in the init messages
  2) editing some bits in the text headers
  3) some specific character combination in the text itself
  (Text IS scrolling twice after pressing source/sat display/ and choosing title/artist etc on the HU)

  By Thomas Landahl, 2018-10-16
  https://github.com/visualapproach/Volvo-melbus

  Comm sniffing and some code contribution by Vincent Gijsen. Thanks a LOT!

*/

#include <EEPROM.h>

#define INT_NUM (byte)1         //Interrupt number (0/1 on ATMega 328P)
#define MELBUS_CLOCKBIT (byte)3 //Pin D2  - CLK
#define MELBUS_DATA (byte)4     //Pin D3  - Data
#define MELBUS_BUSY (byte)5     //Pin D4  - Busy

/*const byte mutePin = 9;
  const byte playPin = 10;
  const byte prevPin = 11;
  const byte nextPin = 12;
  const byte upPin = 7;    //volume up (not used on PCB)
  const byte downPin = 8;  //volume down (not used on PCB)

  const byte LED_RED_LFT = 14;  //A0
  const byte LED_GRN_LFT = 15;  //A1
  const byte LED_BLU_LFT = 16;  //A2
  const byte LED_RED_RGT = 19;  //A3
  const byte LED_GRN_RGT = 18;  //A4
  const byte LED_BLU_RGT = 17;  //A5
*/
const byte MISC = 20;         //A6
const byte BATTERY = 21;      //A7 - PCB uses this
/*
  byte lftClr;
  byte rgtClr;
  String colors[8] = {
  "Black ", "Blue  ", "Green ", "Cyan  ", "Red   ", "Purple", "Yellow", "White "
  };
*/
const int R_CAR = 10; //cable resistance from battery to device
const int R_1 = 3300; //r1 of voltage divider (connected to GND)
const int R_2 = 6800; //r2 of voltage divider (connected to RAW)
const float T_LOSS = 0.8; //if circuit is protected with a transistor and diode, put total voltage drop here.

byte track = 0x01; //Display show HEX value, not DEC. (A-F not "allowed")
byte cd = 0x01; //1-10 is allowed (in HEX. 0A-0F and 1A-1F is not allowed)

//volatile variables used inside AND outside of ISP
volatile byte melbus_ReceivedByte = 0;
volatile byte melbus_Bitposition = 7;
volatile bool byteIsRead = false;

byte byteToSend = 0;          //global to avoid unnecessary overhead
[... 1275 lines cut ...]
}

void SendTrackInfo() {
  for (byte i = 0; i < 9; i++) {
    byteToSend = trackInfo[i];
    SendByteToMelbus();
  }
}

void SendCartridgeInfo() {
  for (byte i = 0; i < 6; i++) {
    byteToSend = cartridgeInfo[i];
    SendByteToMelbus();
  }
}

float getBatV() {
  float a, bv;

  a = analogRead(BATTERY) * 5.0 / 1023.0;
  bv = a * (R_1 + R_2 + R_CAR) / R_1 + T_LOSS;
  return bv;
}

//Happy listening AND READING, hacker!`

Arroquw, 2023-12-08:
Which BT module was used for this?

assa62, 2023-12-09:
BT201