visualapproach/Volvo-melbus #27: HU buttons

Arroquw, 2020-02-14:
Hello, 

Is it possible to receive the next and prev buttons from the HU with the melbus protocol? I'm planning to use a different bluetooth module (https://www.sunrom.com/get/996418), and it's possible to change tracks based on impedance with a button press, so I'm wondering if it's possible to hook up some resistors to arduino pins and write high to a digital pin if a button press if received.

visualapproach, 2020-02-14:
Gnd on DIN shell!
And yes you can use prev/next. It’s already  in the code (my code in this repository). You may need/want to adjust pin numbers in hw and/or sw and adapt to you specific build ofc. Good luck

Arroquw, 2020-02-14:
OH I kept thinking the 4066 components were physical buttons, but they are actually switch IC's. English is not my native language so I didn't know what bilateral means and assumed they meant something along the lines of physical. So D10-12 are used for play/next/prev buttons on the HU. I see. 

By the way, what are the 100R resistances on CLK/BSY/DATA used for?

visualapproach, 2020-02-14:
Yes you are right about the 4066. Those resistors is not my idea, they seem to be ”good practice ”. They attenuate reflections probably.

Arroquw, 2020-02-16:
Alright. What kind of ground loop isolator should I use? The kind with 2 RCA connectors?

visualapproach, 2020-02-17:
That is up to you. I preferred small connectors but it is not important.

closed 2020-04-06 by Arroquw