visualapproach/Volvo-melbus #28: popping sound

Arroquw, 2020-04-06:
Hello,

Currently instead of having a BT module connected, I have an AUX connector for testing. However I hear a constant popping sound, and if I connect the aux cable then it will be attenuated a bit. Ground loop isolator does not fix it. How can I fix this?

visualapproach, 2020-04-06:
I take it you’re not into pop;)

visualapproach, 2020-04-06:
Sorry, but I don’t know. Is it bursts of noise in equally spaced intervals?

Arroquw, 2020-04-07:
Yeah that's it

visualapproach, 2020-04-07:
Do you hear it whilst playing music?

Arroquw, 2020-04-07:
Yes, I hear it always. If nothing is connected, I hear it. If I dont play any music, I hear it. https://github.com/Arroquw/123 here's a video of the sound

visualapproach, 2020-04-07:
Ok, thanks for the video. It was some time now since I did this but it sound like the data signals coming through. Either by galvanic connection (a short somewhere) or by induction. There should be some comments regarding this on this page. Make sure you have only one GND wire connected to the arduino, to avoid a big loop antenna. Also make sure audio common return is not connected to GND.

visualapproach, 2020-04-07:
You could try this also:
https://github.com/visualapproach/Volvo-melbus/issues/19#issuecomment-426168662

Arroquw, 2020-04-07:
I tried the 4k7 resistors, no change at all. common return is not connected to GND either, I've tested that with a multimeter. There is also just the line from the transistor to the arduino as ground (you can see it in the video, it's the tin line going between the arduino pins). My friend also has a volvo and I've made this for him as well, but the board looks a lot different, he has the same problem but the popping is a bit louder.

Arroquw, 2020-04-10:
would a 10uF cap do anything useful on the audio common return or maybe L and r?

visualapproach, 2020-04-10:
Maybe a little but at the cost of also filtrering wanted audio. I Wish I could give you a solution but I can’t. People having this issue are apparantly using cheap BT modules or aux. I had noise too until switching to the CSR module. No guarantee given though. The noise is coming from data signals on melbus. Even of you disconnect your circuit you Will hear it because the HU itself is transmitting on the lines.

assa62, 2020-04-10:
> The noise is coming from data signals on melbus. Even of you disconnect your circuit you Will hear it because the HU itself is transmitting on the lines.

And there is. He's right.

Arroquw, 2020-04-10:
Is there a theory or explanation as to why the CSR fixes the noise?

assa62, 2020-04-11:
> Is there a theory or explanation as to why the CSR fixes the noise?

I think this is an arduino board problem. There are clicks even without a connected module.

visualapproach, 2020-04-11:
It’s clicking even without the arduino if you remove it after the audio input is opened.
Not a theory but some speculation: when nothing is connected the audio input is high impedance and high gain, meaning it will easily pick up inductive noise like data signals. Somehow the CSR manages to overcome that. Possibly by active noise cancelling or strong output?

Arroquw, 2023-11-06:
I can confirm that using a headphone amplifier completely negates the popping noise, likely the internal amplifier of the BT module achieves the same result.