visualapproach/Volvo-melbus #30: Can only select CD03/CD04

JamieEC, 2022-01-01:
I am using the melbus_v7i.ino sketch and my CD player only allows me to select cd 3 or 4. Not having the issue when its not connected. I have tried using the newer firmware but it has a lot of stuff I don't need, the older one seems to be much simpler. Any idea what could be causing this? Could you help me create a cut down version of the newer code with just the basics? Thanks!

JamieEC, 2022-01-01:
Also using the most recent code using the CD player is very glitchy, with timestamps incorrect and sporadic errors displayed.

visualapproach, 2022-01-01:
It worked for me. Don't know what is causing it for you.

JamieEC, 2022-01-01:
What model head unit have you tested with?

JamieEC, 2022-01-02:
Upon investigating, changing the cartridge info to {0x00, 0x0f, 0xFF, 0x4A, 0xFC, 0xFF} seems to have sorted this. I will do some testing and update.