visualapproach/Volvo-melbus #32: Working code with text and buttons

pld360, 2023-05-25:
I recently bought an S60 with the HU-603 radio. I want to make an emulator for bluetooth. I would like for it to support text (so work as MD Changer as my radio doesn't have SAT source) and also controlling the bluetooth device  (next/prev steering wheel or radio buttons). Is there any working ready-made code and schematic that will make this happen? I am a total noob with coding so just looking for something that works and won't fry the radio. Thanks!

VincentGijsen, 2023-11-04:
today we sniffed the MD-changer logic in @https://github.com/Arroquw, Volvo HU803 (similar to 603), car with my old aftermarket emulator iMIV.

raw dumps can be found here: https://github.com/VincentGijsen/melbus-sniffed-traffic/tree/master/md

if you want to take a turn reversing :), or wait until https://github.com/Arroquw has managed to patch the SAT code for those models.

Arroquw, 2023-11-06:
working on my STM32 version first, will try to port the implementation of text once I get that working

Arroquw, 2023-12-08:
Sending text as MD CHGR working on my stm32, just need to add bluetooth now. 

https://github.com/Arroquw/volvo_melbus_stm32

Arroquw, 2023-12-09:
I ported my MD CHGR code to this project: https://github.com/visualapproach/Volvo-melbus/pull/33

Don't think the CSR bluetooth module has text/serial support though